bluesky/ophyd-async #792: Add walk_config_signals so that settings may be stored for config signals only

burkeds, 2025-02-26:
The `store_settings` plan stub walks through all RW signals and stores their values. For Tango devices which have auto-populating signals, this results in a huge number of signals being stored which can include signals like "Position" which, if restored, will move the motor or it can include SignalRWs which are actually Tango commands on the backend. If you recall, this is a compromise until typed SignalXs are possible.

It would be useful if there was a variation of store_settings which used `walk_config_signals` which only saves the config signals of the device, i.e. those returned by `StandardReadable.read_configuration` or those which have `Format.CONFIG_SIGNAL`.

coretl, 2025-02-27:
That sounds reasonable in light of #779. Are commands really `SignalRW`s? I would have thought `SignalW` would be better?

burkeds, 2025-02-28:
> That sounds reasonable in light of [#779](https://github.com/bluesky/ophyd-async/issues/779). Are commands really `SignalRW`s? I would have thought `SignalW` would be better?

I considered that but it causes problem for certain commands. In my opinion, the difference between attributes and commands in Tango is poorly defined. But attributes and commands are capable of receiving input data, performing an action when written, and returning data, potentially of a different type than the input type. This means that commands need to be readable.

closed 2025-02-28 by burkeds