bluesky/ophyd-async #811: Feat: UI component for visualising timing pulses

DiamondJoseph, 2025-03-11:
Equivalent to the ScanSpec visualiser, or the StaticScan preview window for GDA, it would be desirable to be able to plot a TriggerInfo (showing relative live/dead time/number of repetitions?) or the output of a Sequence Table.

Or, do we have any plans for something similar? e.g. visualising a ScanSpec in the DURATION axis?

This was requested by Richard Dixey on behalf of i22 

## Acceptance Criteria
- Specific criteria that will be used to judge if the issue is fixed

coretl, 2025-03-11:
I think this becomes:
- Remove `get_deadtime` from `StandardDetector`
- Make a `DeadtimeProvider` that can be used at the plan level to always pass the deadtime in
- Use that `DeadtimeProvider` in a top level service so we can plot these UIs in a sync way

jwlodek, 2025-03-11:
+1 to making fetching deadtime the responsibility of a provider

DiamondJoseph, 2025-03-17:
@RJCD-Diamond