bluesky/ophyd-async #820: Docs build broken

coretl, 2025-03-13:
We get:
```
/home/runner/work/ophyd-async/ophyd-async/docs/_api/ophyd_async/ophyd_async.epics.adkinetix.md:95: WARNING: py:class reference target not found: ophyd_async.epics.adkinetix._kinetix_io.KinetixReadoutMode [ref.class]
/home/runner/work/ophyd-async/ophyd-async/docs/_api/ophyd_async/ophyd_async.epics.advimba.md:92: WARNING: py:class reference target not found: ophyd_async.epics.advimba._vimba_io.VimbaConvertFormat [ref.class]
/home/runner/work/ophyd-async/ophyd-async/docs/_api/ophyd_async/ophyd_async.epics.advimba.md:125: WARNING: py:class reference target not found: ophyd_async.epics.advimba._vimba_io.VimbaOverlap [ref.class]
```

## Steps To Reproduce
Steps to reproduce the behavior:
1. Run `tox -e docs`
4. See error

coretl, 2025-03-13:
@jwlodek any ideas?

jwlodek, 2025-03-13:
Let me investigate...

coretl, 2025-03-14:
Fixed by #822, thanks!

closed 2025-03-14 by coretl