bluesky/ophyd-async #821: AD Writer does not recognize NDAttributes from the detector driver itself

canismarko, 2025-03-13:
Seems like the ADHDFWriter assumes that only plugins will specify ndattributes that get sent to the HDF file. This is not true for one of our classes of detectors and so our ndattributes are not getting into the descriptor document.

We have an XSpress3 detector. The EPICS driver in the IOC [adds several parameters](https://github.com/epics-modules/xspress3/blob/36e9c4143807d452d9d8441a769285b24bdeb838/xspress3App/src/xspress3Epics.cpp#L1809) for each element for tracking dead time and other similar properties.

To reconcile this with ophyd-async, I wrote XML for these attributes to the NDAttributesFile PV on the detector itself (e.g. ``detector.driver.nd_attributes_file``). In ophyd_async <0.9, this worked as intended. I can look at the descriptor and see these attributes listed as data keys. These attributes are not coming from any of the 

Now with ophyd-async 0.9.0, the ADHDFWriter [looks in it's `plugins` dictionary](https://github.com/bluesky/ophyd-async/blob/ee56ab71521ce555803a73aaddbae53e9d17c116/src/ophyd_async/epics/adcore/_hdf_writer.py#L90) to find NDAttributes. Since the driver is not listed in this dictionary, the attributes are not recognized.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Create an ophyd-async device for an area detector
2. Write XML for NDAttributes to ``detector.drier.nd_attributes_file``
3. Generate descriptors for the detector: ``await detector.describe()``
4. Note that the ndattributes are not included in the description.

## Acceptance Criteria
- Attributes specified in ``detector.driver.nd_attributes_file`` show up in ``await detector.describe()``.

## Workarounds

### Option 1: Add the detector to ``writer._plugins``

This has to be done after initialization. Using ``Writer(plugins={"driver": driver, **plugins}, ...)`` will add the driver to plugins, but also renames the detector for some reason, not entirely clear why.

Instead, after creating the writer, we can do ``detector._writer._plugins['camera'] = detector.driver``. This produces the desired results. However, this fails type checking since plugins is of type ``dict[str, NDPluginBaseIO]`` and the driver is not of type ``NDPluginBaseIO``.

### Option2: Use a dummy plugin just for the ndattributes

I haven't tested this, but I suspect I could pick a plugin that doesn't modify the underlying NDArray, but let's me add the XML to the NDAttributesFile. Then I would write an NDPluginBaseIO subclass and add this to the ``plugins`` dictionary. This seems kludgy to me since we're using a plugin for something unrelated to its purpose, and for something that can easily be handled in EPICS by the driver itself.

coretl, 2025-03-17:
I would go for option number 3, pass the `driver` to https://github.com/bluesky/ophyd-async/blob/93ac49901f6b32e5a8243a278119c9ced53dc7b4/src/ophyd_async/epics/adcore/_core_writer.py#L68-L75 as an extra argument

canismarko, 2025-03-19:
> I would go for option number 3, pass the driver to 

@coretl You're proposing this as a change to ophyd_async? Happy to prepare that PR.