bluesky/ophyd-async #825: StandardDetector.stage behavior with writer and trigger info

thomashopkins32, 2025-03-17:
Hello, while implementing #726, I ran into an issue where you can't really set the `TriggerInfo` for a simple `count` plan easily.

For example, if I try to do
```python
trigger_info = TriggerInfo(number_of_events=1, exposures_per_event=5)
# Prepare the detector to use capture 5 exposures in a single event
RE(bps.prepare(det, trigger_info))
# EXPECTED: Gather data for 5 events. 5 events x 5 exposures per event = 25 total
# ACTUAL: 5 total since trigger_info is unset in stage()
RE(bp.count([det], num=5))
```

The only way around this is to create a custom plan that calls `stage()` before `bps.prepare` and make it into a single RE call.

This begs the question, should we really be unsetting `TriggerInfo` during `stage()`?
Should we be closing the file writer during `stage()`?

https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/core/_detector.py#L230-L231

## Acceptance Criteria
- My example above runs and produces 25 exposures as output

coretl, 2025-03-17:
Hmm, I'm not sure about this. We use `stage` at the moment to say "put into a known good state", which should close the file writer ready for a new file, but I'm not sure whether it should unset the `TriggerInfo`.

- Pro: There is prior art in that `TriggerInfo(livetime=None)` means "take the last configured exposure from the detector", which will also survive `stage`
- Con: If you `prepare` once to try things out, then someone else goes down to run the `count` plan, it might be surprising that it now does something different, especially as there's nothing corresponding to `exposures_per_event` in EPICS to inspect

@danielballan @tacaswell @jwlodek @mrakitin thoughts?

An alternative would be to add support for `prepare` to `count` in someway, either with an extra `prepare` argument, or a custom `per_shot` plan...

thomashopkins32, 2025-03-18:
> so we wouldn't want to read that back for use as `exposures_per_event` for the next step scan...

For step scans I think we have to be able to read it back from EPICS. If someone manually configures the `self.driver.num_images` they would expect that value to stick around. I see why this can't be the case for a fly scan.

For example,

```python
def custom_count(det, exposures_per_event = 5):
    # Set the exposures_per_event at the EPICS level
    yield from bps.mv(det.controller.driver.num_images, exposures_per_event)
    # EXPECTED: Collect `exposures_per_event` number of images for a single event
    # ACTUAL: Always collects 1 image for a single event
    yield from bp.count([det])

RE(custom_count(det))
```
The above example currently doesn't work as I would expect. In the `bp.count`, `det.trigger` calls the `det.prepare` method is called and resets the `det.controller.driver.num_images` to 1 before collecting images. You would have to use `bps.prepare` in place of `bps.mv` for this to work because it would cache the `TriggerInfo` in the detector.

Should we use `TriggerInfo` for step scans at all? Maybe we can avoid these problems by separating the logic a bit more.

thomashopkins32, 2025-03-24:
Following up on our discussion in the bluesky call today, I think that we should try our best to respect the current EPICS configuration as the default behavior. From a user's perspective, it can be convenient for diagnostic and small experiments to do some configuration work manually (like setting `self.driver.num_images` in CS Studio), then running a standard bluesky plan.

I foresee it being frustrating for users to see their configuration wiped out when they run a standard bluesky plan like `count`.

ophyd sync tends to repsect the current EPICS configuration in my experience (definitely for the equivalent of `self.driver.num_images`).

coretl, 2025-03-24:
> I foresee it being frustrating for users to see their configuration wiped out when they run a standard bluesky plan like `count`.

As far as I can see we have two options:
1. Make `trigger()` without `prepare()` use `num_images` from EPICS. This means that if you use the detector in a flyscan for 1000 exposures, then run a `count` plan, it will take 1000 exposures per event. If you want to ensure `count` always works the same way, you need to wrap it in a custom plan, or call `bps.mv(det.drv.num_images, 1)` before.
2. Make `trigger()` without `prepare()` override `num_images` from EPICS. This means that no matter what state your detector is in, it will always take 1 exposure per event. If you want to do `count` with a different number of exposures per event, you need to wrap it in a custom plan.

Personally I favour 2. I think using a single EPICS field as the source of `exposures_per_event` and the destination of `total_number_of_exposures` is fragile, and will lead to more surprises and frustrations at unpredictable times...

> What this means in practice is up for discussion. I might be in favor of making `TriggerInfo` + `prepare` optional altogether unless there is a real requirement I am missing.

I'm not sure what you mean, isn't it already optional? You can `stage`, then `trigger` without needing `prepare` in between. It will respect all EPICS variables apart from `num_images` which it always overwrites...