bluesky/ophyd-async #829: Add signals for "unhappy path" of `DerivedSignal`

coretl, 2025-03-21:
#661 introduced `DerivedSignal`, but there are quite a few code paths untested, like:
https://github.com/bluesky/ophyd-async/blob/e16843c0a19d335409dc640dfe71df1a3228a08c/src/ophyd_async/core/_derived_signal_backend.py#L199-L201

https://app.codecov.io/gh/bluesky/ophyd-async/blob/derived-signal/src%2Fophyd_async%2Fcore%2F_derived_signal_backend.py#L200

These should all be covered, either by writing new tests or extending the existing tests

## Acceptance Criteria
- Test suite extended to cover these branches.

Villtord, 2025-03-31:
I can try to do this if nobody works on it yet

coretl, 2025-03-31:
> I can try to do this if nobody works on it yet

yes please!

Villtord, 2025-03-31:
Sorry I am confused - is this function really supposed to throw TypeError whenever there is a mismatch in input list? This doesn't sound like a filter I am used to.
```
def filter_by_type(raw_devices: Mapping[str, Any], type_: type[T]) -> dict[str, T]:
    filtered_devices: dict[str, T] = {}
    for name, device in raw_devices.items():
        if not isinstance(device, type_):
            msg = f"{device} is not an instance of {type_}"
            raise TypeError(msg)
        filtered_devices[name] = device
    return filtered_devices
```

coretl, 2025-03-31:
> Sorry I am confused - is this function really supposed to throw TypeError whenever there is a mismatch in input list? This doesn't sound like a filter I am used to.
>
> ```
> def [...]

The function should probably be named `raise_if_not_type`...

thomashopkins32, 2025-03-31:
@Villtord can you rename your branch on the remote to exclude `"` characters? I think it may be breaking Windows CI: https://github.com/bluesky/ophyd-async/actions/runs/14177724812/job/39716759349?pr=726#step:2:63

Villtord, 2025-04-01:
Sorry I realised it later when I saw broken CI, renamed branch.

coretl, 2025-06-11:
@Villtord is this done?