bluesky/ophyd-async #841: Tetramm freezes when preparing for flyscan

RJCD-Diamond, 2025-03-26:
On i22 there are several Tetramms that need to be hardware triggered by the PandAbox. This code works when preparing the saxs/waxs Pilatus detectors but hangs when preparing the Tetramm's (I0 or It)), before timing out. After timing out it leaves the Tetramm IOC epics IOC in a faulty state, and requires resetting to restore normal functionality.

The code below should be a minimal example in order to reproduce the issue. Running on latest version of bluesky, and ophyd-async 0.9.0a2.

The code was run where detectors contained the i22 SAXS Pilatus and the 22 it Tetramm or the 22 i0 Tetramm, imported from Dodal

```python
from bluesky.run_engine import RunEngine
import bluesky.plan_stubs as bps

from ophyd_async.core import (
    DetectorTrigger,
    StandardDetector,
    StandardFlyer,
    TriggerInfo)

from ophyd_async.fastcs.panda import (
    HDFPanda,
    SeqTable,
    SeqTableInfo,
    StaticSeqTableTriggerLogic,
)

from ophyd_async.plan_stubs import fly_and_collect

def prepare_flyscan(panda: HDFPanda, seq_table: SeqTable, detectors: list[StandardDetector]):
    
    table_info = SeqTableInfo(sequence_table=seq_table, repeats=1)

    trigger_info = TriggerInfo(number_of_triggers=1, 
                            trigger=DetectorTrigger.CONSTANT_GATE, 
                            deadtime=4e-5,
                            livetime=1,
                            multiplier=1,
                            frame_timeout=None)

    trigger_logic = StaticSeqTableTriggerLogic(panda.seq[1])
    flyer = StandardFlyer(trigger_logic)

    yield from bps.stage_all(*detectors, flyer, group="prepare")
    yield from bps.prepare(flyer, table_info, wait=False, group="prepare")

    for det in detectors:
        yield from bps.prepare(det, trigger_info, wait=False, group="prepare")
    
    yield from bps.wait(group="prepare", timeout=30)

    yield from fly_and_collect(
            stream_name='primary',
            detectors=detectors,
            flyer=flyer,
        )

```

## Acceptance Criteria
- Prepare Tetramm's triggered by a PandAbox, without hanging 
- run: yield from fly_and_collect(stream_name='primary', detectors=detectors, flyer=flyer)
- Save Tetramm data

coretl, 2025-03-26:
@barnettwilliam do you have any time to investigate this? It would be good to narrow down further which particular PV operations puts the Tetramm in the faulty state...

RJCD-Diamond, 2025-03-27:
Also fyi, it produces the error on timeout:

```
future: <Task finished name='Task-4286' coro=<StandardFlyer.unstage() done, defined at /home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_flyer.py:52> exception=NameError("name '_context' is not defined")>
Traceback (most recent call last):
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_flyer.py", line 54, in unstage
    await self._trigger_logic.stop()
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/fastcs/panda/_trigger.py", line 41, in stop
    await wait_for_value(self.seq.active, False, timeout=1)
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 490, in wait_for_value
    await checker.wait_for_value(signal, timeout)
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 450, in wait_for_value
    await asyncio.wait_for(self._wait_for_value(signal), timeout)
  File "/dls_sw/apps/python/miniforge/4.10.0-0/envs/python3.11/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 441, in _wait_for_value
    async for value in observe_value(signal):
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 346, in observe_value
    async for _, value in observe_signals_value(
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 407, in observe_signals_value
    signal.subscribe_value(queue_value)
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 210, in subscribe_value
    self._get_cache().subscribe(function, want_value=True)
    ^^^^^^^^^^^^^^^^^
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 183, in _get_cache
    self._cache = _SignalCache(self._connector.backend, self)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/core/_signal.py", line 108, in __init__
    backend.set_callback(self._callback)
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/epics/core/_p4p.py", line 424, in set_callback
    self.subscription = context().monitor(
                        ^^^^^^^^^
  File "/home/akz63626/i22/i22_venv/lib/python3.11/site-packages/ophyd_async/epics/core/_p4p.py", line 308, in context
    if _context is None:
       ^^^^^^^^
NameError: name '_context' is not defined
```

coretl, 2025-03-27:
That looks like a bug, but it looks like it is caused by `unstage` being called after the `atexit` call. It shouldn't make any difference to this ticket however.

coretl, 2025-03-27:
Also, does the error still happen when you connect over CA instead of PVA?

RJCD-Diamond, 2025-03-27:
Sorry don't know. How would I connect over CA instead of PVA?

barnettwilliam, 2025-03-27:
No time immediately, I may be have a look when I start testing the pressure jump cell again but that won't be until the end of next week at the earliest

coretl, 2025-03-28:
> Sorry don't know. How would I connect over CA instead of PVA?

In the dodal device instance, instead of prefix `pva://BLxxI-EA-TTRM-01`, use `ca://BLxxI-EA-TTRM-01`

barnettwilliam, 2025-05-01:
Testing in lab p38 using dodal 1.44.1 and Tetramm XBPM2 i1 serial no 17013, I have not been able to get the Tetramm into an invalid state where it freezes after `prepare` timing out. The Tetramm behaves as expected with the prepare value being set and continuing to respond during subsequent manual acquisitions.

The Tetramm freezing could potentially be a device specific issue that I will need test at i22 on a machine day.

I do see also see the wait for prepare group timing out which I'm not entirely sure why as there a no significant differences to what is done in the linkam dodal device's `prepare()` compared to the `PilatusDetector`  used in saxs. I also get a different error on time out `WaitForTimeoutError` but this could be because I am running the plan via blueapi and not locally.

##### Tetramm PVs modified during prepare

```
[t:11] BL38P-EA-XBPM-02:DRV:Acquire.VAL Done -> Done 
[t:11] BL38P-EA-XBPM-02:HDF5:Capture.VAL Done -> Done 
[t:11] BL38P-EA-XBPM-02:HDF5:ChunkSizeAuto.VAL Yes -> Yes 
[t:11] BL38P-EA-XBPM-02:HDF5:NumExtraDims.VAL 0 -> 0 
[t:11] BL38P-EA-XBPM-02:HDF5:LazyOpen.VAL Yes -> Yes 
[t:11] BL38P-EA-XBPM-02:HDF5:SWMRMode.VAL On -> On 
[t:11] BL38P-EA-XBPM-02:HDF5:XMLFileName.VAL 0 -> 0 
[t:11] BL38P-EA-XBPM-02:HDF5:EnableCallbacks.VAL Enable -> Enable 
[t:11] BL38P-EA-XBPM-02:HDF5:CreateDirectory.VAL 0 -> 0 
[t:11] BL38P-EA-XBPM-02:HDF5:FilePath.VAL [47, 100, 108, 115, 47, 112, 51, 56, 47, 100, 97, 116, 97, 47, 50, 48, 50, 52, 47, 99, 109, 51, 55, 50, 56, 50, 45, 50, 47, 98, 108, 117, 101, 115, 107, 121, 0] -> [47, 100, 108, 115, 47, 112, 51, 56, 47, 100, 97, 116, 97, 47, 50, 48, 50, 52, 47, 99, 109, 51, 55, 50, 56, 50, 45, 50, 47, 98, 108, 117, 101, 115, 107, 121, 0] 
[t:11] BL38P-EA-XBPM-02:HDF5:FileName.VAL [112, 51, 56, 45, 49, 45, 105, 49, 0] -> [112, 51, 56, 45, 49, 45, 105, 49, 0] 
[t:11] BL38P-EA-XBPM-02:HDF5:FileWriteMode.VAL Stream -> Stream 
[t:11] BL38P-EA-XBPM-02:HDF5:FileTemplate.VAL [37, 115, 37, 115, 46, 104, 53, 0] -> [37, 115, 37, 115, 46, 104, 53, 0] 
[t:11] BL38P-EA-XBPM-02:HDF5:AutoIncrement.VAL Yes -> Yes 
[t:11] BL38P-EA-XBPM-02:HDF5:FileNumber.VAL 0 -> 0 
[t:11] BL38P-EA-XBPM-02:HDF5:NumCapture.VAL 0 -> 0 
[t:11] BL38P-EA-XBPM-02:DRV:TriggerMode.VAL Free run -> Ext. trig. 
[t:11] BL38P-EA-XBPM-02:DRV:AveragingTime.VAL 0.1 -> 1 
[t:11] BL38P-EA-XBPM-02:DRV:ValuesPerRead.VAL 5 -> 100 
[t:41] BL38P-EA-XBPM-02:DRV:Acquire.VAL Acquire -> Done 
[t:41] BL38P-EA-XBPM-02:HDF5:Capture.VAL Capture -> Done 
[t:41] BL38P-EA-XBPM-02:HDF5:Capture.VAL Done -> Done 
[t:41] BL38P-EA-XBPM-02:DRV:Acquire.VAL Done -> Done 
```

##### Error on timeout

```
Traceback (most recent call last): 
  File "/venv/lib/python3.11/site-packages/bluesky/run_engine.py", line 1584, in _run 
    msg = self._plan_stack[-1].throw(stashed_exception or resp) 
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/utils/__init__.py", line 1258, in dec_inner 
    return (yield from plan) 
            ^^^^^^^^^^^^^^^ 
  File "/dls_sw/p38/software/blueapi/scratch/dodal/src/dodal/plan_stubs/data_session.py", line 43, in attach_data_session_metadata_wrapper 
    yield from bpp.inject_md_wrapper(plan, md={DATA_SESSION: data_session}) 
  File "/venv/lib/python3.11/site-packages/bluesky/preprocessors.py", line 755, in inject_md_wrapper 
    return (yield from msg_mutator(plan, _inject_md)) 
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/preprocessors.py", line 265, in msg_mutator 
    msg = plan.throw(_e) 
          ^^^^^^^^^^^^^^ 
  File "/dls_sw/p38/software/blueapi/scratch/i22_bluesky/src/i22_bluesky/plans/test_p38_devices.py", line 109, in test_p38_tetramm_prepare 
    yield from bps.wait(group="prepare", timeout=30) 
  File "/venv/lib/python3.11/site-packages/bluesky/utils/__init__.py", line 1952, in __iter__ 
    return (yield from self._iter) 
            ^^^^^^^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/plan_stubs.py", line 645, in wait 
    return (yield Msg("wait", None, group=group, error_on_timeout=error_on_timeout, timeout=timeout)) 
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/preprocessors.py", line 259, in msg_mutator 
    _s = yield msg 
         ^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/run_engine.py", line 1676, in _run 
    new_response = await coro(msg) 
                   ^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/opentelemetry/util/_decorator.py", line 71, in async_wrapper 
    return await func(*args, **kwargs)  # type: ignore 
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
  File "/venv/lib/python3.11/site-packages/bluesky/run_engine.py", line 2359, in _wait 
    await self._wait_for(Msg("wait_for", None, futs, timeout=msg.kwargs.get("timeout", None))) 
  File "/venv/lib/python3.11/site-packages/bluesky/run_engine.py", line 1824, in _wait_for 
    raise WaitForTimeoutError("Plan failed to complete in the specified time") 

bluesky.run_engine.WaitForTimeoutError: Plan failed to complete in the specified time 

```

barnettwilliam, 2025-05-09:
This may be tetramm dodal device calling async multiple times on `_wait_for_idle`:
https://github.com/DiamondLightSource/dodal/blob/913c0a3c0c6a9aa4f01880cd89098db554ff31b5/src/dodal/devices/tetramm.py#L133-L138

Removing the await on line 134 the prepare timeout behaviour is not seen and the plan completes however it is difficult to verify the actual runtime values using debug logs as dodal device-level messages are not visible. I've raised the following issue here https://github.com/DiamondLightSource/dodal/issues/1221