bluesky/ophyd-async #849: Velocity calculated to zero

stan-dot, 2025-04-03:
Expected behavior - I can set the value easily and the sim motor moves

```python
/workspaces/dodal/tests/devices/unit_tests/test_turbo_slit.py::test_turbo_slit_set failed: Traceback (most recent call last):
  File "/venv/lib/python3.11/site-packages/ophyd_async/epics/motor.py", line 193, in set
    abs((new_position - old_position) / velocity)
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
ZeroDivisionError: float division by zero
```

## Steps To Reproduce
Steps to reproduce the behavior:
1. Go to the PR for turbo slit https://github.com/DiamondLightSource/dodal/pull/1126/files
2. clone to local machine
3. Go to `test_turbo_slit.py`
4. Run the test

``` python

@pytest.fixture
def slit(RE) -> TurboSlit:
    with init_devices(mock=True):
        slit = TurboSlit(prefix="TEST-EA-TURBOSLIT:", name="turbo_slit")
    return slit

async def test_turbo_slit_set(slit: TurboSlit):
    set_mock_value(slit.xfine.user_readback, 0.0)
    await slit.set(0.5) # error at this line
    raise AssertionError("This test is not implemented yet")
```

## Acceptance Criteria
- test progresses to the next line

coretl, 2025-04-03:
The velocity is set to zero, this has undefined behaviour on the EPICS side. It might mean "move as fast as you can", or it might mean "sit there forever until someone else changes the velocity on the controller". The appropriate fix is probably to improve the error message to say to set a non-zero velocity. The fix for your test is to `set_mock_value(slit.velocity, 1)` in the fixture.

Do you agree with the fix? If so, would you mind adding a PR that improves the error message?

stan-dot, 2025-04-03:
hmm, okay I didn't expect to need to set velocity, but it makes sense. I will add velocity, thanks I appreciate the tip

stan-dot, 2025-04-03:
yay it works now, thanks!

closed 2025-04-03 by stan-dot