bluesky/ophyd-async #855: RuntimeError: dictionary changed size during iteration

xkronosua, 2025-04-09:
For loop over `self._listeners.items()` in `_callback` method can lead to `RuntimeError: dictionary changed size during iteration` when we add a new listener during plan execution in RunEngine. It stops further callbacks in the plan.

```
Traceback (most recent call last):
  File "...\p4p\client\asyncio.py", line 383, in _handle
    await self._cb(E)
  File "...\ophyd_async\epics\core\_p4p.py", line 427, in async_callback
    callback(self._make_reading(v))
    ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "...\ophyd_async\core\_signal.py", line 143, in _callback
    for function, want_value in self._listeners.items():
                                ~~~~~~~~~~~~~~~~~~~~~^^
RuntimeError: dictionary changed size during iteration
```

Proposed solution:

```
def _callback(self, reading: Reading[SignalDatatypeT]) -> None:
        self._signal.log.debug(
            f"Updated subscription: reading of source {self._signal.source} changed "
            f"from {self._reading} to {reading}"
        )
        self._reading = reading
        self._valid.set()
        items = self._listeners.copy().items()
        for function, want_value in items:
            self._notify(function, want_value)
```

ophyd-async was built from main repository.
Version: 0.9.0a3.dev47+g2984ea49

coretl, 2025-04-09:
Good catch! Do you have an example plan that provokes this that we can turn into a unit test for this?

xkronosua, 2025-04-09:
I tried to make a simplified plan to demonstrate the error, but without success. It appears randomly when I work with many devices controlled in multi-run plans.

coretl, 2025-04-11:
Would you be able to post the full plan? It might give a clue as to where it fails...

closed 2025-07-01 by coretl