bluesky/ophyd-async #867: `derived_signal_rw` cannot derived from a read-only signal

canismarko, 2025-04-24:
I'm testing out the new `derived_signal_rw` from v0.10.0a2, and I think I've run into an edge case.

It looks like `derived_signal_rw` checks that every raw signal is a Locatable. The problem is that some of the raw signals are read-only. I need these signals to calculate `raw_to_derived()`, but I cannot include them since they are not Locatable. This should conceptually be okay since they are read in `set_derived()`, but not set.

Here's my actual use-case, with a MWE below.

I have a device that represents an asymmetric crystal analyzer. There's a bunch of hairy math involved so I put it all into [a device](https://github.com/spc-group/haven/blob/13433a65872a66928ab664ac5d36e03e8b9c03c6/src/haven/devices/asymmotron.py#L202) with derived signals and a bunch of [parameterized tests](https://github.com/spc-group/haven/blob/energy_devices/src/haven/tests/test_asymmotron.py).

I have the following structure. `Analyzer` is the top-level device.

`Analyzer.d_spacing` is a `derived_signal_r` that derives from:
- `Analyzer.reflection.h`
- `Analyzer.reflection.k`
- `Analyzer.reflection.l`
- `Analyzer.lattice_constant`

I think this signal should be read-only since `d_spacing -> (h, k, l)` is under-determined, meaning there's no clear way to set the d-spacing directly, you just need to set (h, k, l) individually. (NB: as a workaround in the linked example, `Analyzer.d_spacing` is a `derived_signal_rw` so I can get it to work, but it just raises an exception if you try to set it.)

Then I have the [`Analyzer.energy.setpoint`](https://github.com/spc-group/haven/blob/13433a65872a66928ab664ac5d36e03e8b9c03c6/src/haven/devices/asymmotron.py#L451) derived_signal_rw, which derives from:
- `Analyzer.rowland_diameter`
- `Analyzer.d_spacing`
- `Analyzer.wedge_angle`
- `Analyzer.asymmetry_angle`
- `Analyzer.horizontal`
- `Analyzer.vertical`

defined as the following, where `xtal` is the parent device. *horizontal* and *vertical* are real motors, and the rest are modal values, used to do the calculations.

```python
self.setpoint = derived_signal_rw(
    raw_to_derived=xtal._to_energy,
    set_derived=xtal._set_from_energy,
    derived_units="eV",
    x=xtal.horizontal.user_setpoint,
    y=xtal.user_setpoint,
    D=xtal.rowland_diameter,
    d=xtal.d_spacing,
    beta=xtal.wedge_angle,
    alpha=xtal.asymmetry_angle,
)
```

I need the value for `xtal.d_spacing` to do the calculation in `xtal._to_energy`, so I have to include it in the arguments since `xtal._to_energy` is not a coroutine. In `xtal._set_from_energy` I read from `xtal.d_spacing`, but only set `xtal.horizontal` and `xtal.vertical`.

## Steps To Reproduce

Steps to reproduce the behavior:
1. Create a read-only signal of some flavor.
2. Create a derived_signal_rw that derives from the signal in step 1.
3. Attempt to use the derived_signal_rw.
4. See error

Here's a MWE:

```python
from functools import partial

from ophyd_async.core import soft_signal_r_and_setter, soft_signal_rw, derived_signal_rw

# Create raw signals
raw_sig_ro, _ = soft_signal_r_and_setter(float)
raw_sig_rw = soft_signal_rw(float)

# Define transformations
# (I know I'm being a bit loose with variable scope here,
#  these would normally be bound instance methods)
def average(ro: float, rw: float) -> float:
    return (ro+rw)/2

async def set_rw(value: float):
    val0 = await raw_signal_ro.get_value()
    new_val = 2*value - val0
    await raw_signal_rw.set(new_val)

# Try and create the derived signal...
derived_sig = derived_signal_rw(
    raw_to_derived=average,
    set_derived=set_rw,
    rw=raw_sig_rw,
    ro=raw_sig_ro,
)

```

This example produces the follow exception:

```python
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[17], line 1
----> 1 derived_sig = derived_signal_rw(
      2     raw_to_derived=average,
      3     set_derived=set_rw,
      4     rw=raw_sig_rw,
      5     ro=raw_sig_ro,
      6 )

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/site-packages/ophyd_async/core/_derived_signal.py:244, in derived_signal_rw(raw_to_derived, set_derived, derived_units, derived_precision, **raw_devices)
    239     raise TypeError(msg)
    241 factory = _make_factory(
    242     raw_to_derived=raw_to_derived, set_derived=set_derived, raw_devices=raw_devices
    243 )
--> 244 return factory.derived_signal_rw(
    245     datatype=raw_to_derived_datatype,
    246     name="value",
    247     units=derived_units,
    248     precision=derived_precision,
    249 )

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/site-packages/ophyd_async/core/_derived_signal.py:137, in DerivedSignalFactory.derived_signal_rw(self, datatype, name, units, precision)
    121 def derived_signal_rw(
    122     self,
    123     datatype: type[SignalDatatypeT],
   (...)
    126     precision: int | None = None,
    127 ) -> SignalRW[SignalDatatypeT]:
    128     """Create a read-write derived signal.
    129
    130     :param datatype: The datatype of the derived signal value
   (...)
    135     :param precision: Number of digits after the decimal place to display
    136     """
--> 137     return self._make_signal(SignalRW, datatype, name, units, precision)

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/site-packages/ophyd_async/core/_derived_signal.py:79, in DerivedSignalFactory._make_signal(self, signal_cls, datatype, name, units, precision)
     77     raise ValueError(msg)
     78 if issubclass(signal_cls, SignalRW):
---> 79     self._transformer.raw_locatables  # noqa: B018
     80 backend = DerivedSignalBackend(
     81     datatype, name, self._transformer, units, precision
     82 )
     83 return signal_cls(backend)

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/functools.py:981, in cached_property.__get__(self, instance, owner)
    979 val = cache.get(self.attrname, _NOT_FOUND)
    980 if val is _NOT_FOUND:
--> 981     val = self.func(instance)
    982     try:
    983         cache[self.attrname] = val

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/site-packages/ophyd_async/core/_derived_signal_backend.py:94, in SignalTransformer.raw_locatables(self)
     92 @cached_property
     93 def raw_locatables(self) -> dict[str, AsyncLocatable]:
---> 94     return filter_by_type(self._raw_devices, AsyncLocatable)

File /home/beams1/WOLFMAN/miniforge3/envs/haven/lib/python3.10/site-packages/ophyd_async/core/_derived_signal_backend.py:69, in filter_by_type(raw_devices, type_)
     67     if not isinstance(device, type_):
     68         msg = f"{device} is not an instance of {type_}"
---> 69         raise TypeError(msg)
     70     filtered_devices[name] = device
     71 return filtered_devices

TypeError: <ophyd_async.core._signal.SignalR object at 0x7f7ad441c7c0> is not an instance of <class 'ophyd_async.core._protocol.AsyncLocatable'>
```

## Acceptance Criteria
- The above MWE builds without exceptions, and the derived signal can set the underlying ``soft_signal_rw``.

canismarko, 2025-04-28:
From developer's call, I'll try and implement this with the many-to-many pattern and see what impact this has on the code.

shihab-dls, 2025-05-12:
Looks like this has been fixed by #852, and supported from v0.10.0a3! If you're happy with this, I'll close the issue.

canismarko, 2025-05-12:
Thanks, @shihab-dls. I will take a look and close the issue if it works.

canismarko, 2025-05-14:
Seems to be working, thanks!

closed 2025-05-14 by canismarko