bluesky/ophyd-async #871: Better documentation or exception raising on not using Reference

DominicOram, 2025-04-28:
Currently if a signal is passed around w/o using `Reference` it may get renamed. This behaviour isn't obvious and likely will lead to insidious bugs. Currently it's not that obvious that `Reference` exists or that it fixes the issue, leading to people not using it (https://github.com/bluesky/ophyd-async/pull/870/files#r2063499820)

We should make this more obvious:
* Could we raise a warning if a signal is getting renamed?
* Could we at least document the use of `Reference` a bit more?

## Acceptance Criteria
- It is more obvious when `Reference` should be used

coretl, 2025-04-28:
In this particular case you don't need `Reference`, because you are making it a child of `DetectorWriter` which is not a `Device` subclass. Did the above case lead to the object getting renamed?

At the moment we catch this with `Device.__setattr__`:
https://github.com/bluesky/ophyd-async/blob/f64dfd010a64f269140c0956ebbe65dc84804ee8/src/ophyd_async/core/_device.py#L137-L145

Maybe this error message should have something about `Reference` in it? Could you think of some better text?

DominicOram, 2025-04-28:
Ah, ok, I was being overly cautious sorry. The existing exception is fine.

closed 2025-04-28 by DominicOram