bluesky/ophyd-async #876: Improve error message on panda firmware version mismatch

callumforrester, 2025-04-30:
Currently if the panda firmware is out of date you get a cryptic error from PVI because it can't find certain blocks that are provided by later versions of firmware. We can improve this error message by:

1. Exposing the firmware version in a PV
2. Setting a required firmware version in the ophyd-async object (`HDFPanda`)
3. Adding the version PV as an optional part of the PVI structure
4. Doing a version check in the connection logic

## Acceptance Criteria
- If the panda firmware is too old, a sensible error message is provided in the connect exception

coretl, 2025-04-30:
On reflection, I think we are better off on relying on the presence of the fields rather than a firmware version.

I would prefer to do the following:
- Set the newly added fields as `Optional[SignalRW[float]]` rather than `SignalRW[float]`
- Add an `on_connect` handler to the flyer logic
- In `on_connect` `if pulse.step is None` then `raise RuntimeError("Pulse block has no step field. This was added in firmware version 2.0 so you may need to update the PandA firmware")`

jwlodek, 2025-04-30:
As of this PR: https://github.com/PandABlocks/PandABlocks-ioc/pull/150 there is a PV for firmware version

coretl, 2025-05-01:
> How does it know that `step` was added in version 2.0?

Because we hard code that in the error string of the first bit of logic that needs it...

On even more reflection, @callumforrester 's original solution is better. How about this:
- Try to connect
- Catch any PVI errors, report an error message with the fields that are mismatching, the current firmware version number, and the required version number