bluesky/ophyd-async #878: Improve error message from _datakey_shape function

oliwenmandiamond, 2025-05-01:
The `_datakey_shape` function located in `ophyd_async.core._signal_backend.py` looks like the following:
```Python
def _datakey_shape(value: SignalDatatype) -> list[int]:
    if type(value) in _primitive_dtype or isinstance(value, EnumTypes):
        return []
    elif isinstance(value, np.ndarray):
        return list(value.shape)
    elif isinstance(value, Sequence | Table):
        return [len(value)]
    else:
        raise TypeError(f"Can't make shape for {value}")
```
The TypeError message could be improved to display the `SignalDatatype` that failed e.g 

`raise TypeError(f"Can't make shape for {value} with SignalDatatype: {type(value}")`

This significantly helps with debugging when you run into this error as you can narrow down which field was causing the error better with the type information as well e.g 

`TypeError: Can't make shape for 40269600.0, with SignalDatatype <class 'numpy.float64'>`

coretl, 2025-05-09:
Good idea, do you fancy making a PR for this?

gkalua, 2025-10-08:
@oliwenmandiamond Could you share an example of the sort of signal that triggers the error? Tried a soft signal and that didnt trigger this error.