bluesky/ophyd-async #881: NDPluginStatsIO ProfileX and ProfileY signals are RW

dan-fernandes, 2025-05-02:
The `NDPluginStatsIO` plugin contains signals ProfileX and ProfileY. These are both read/write in the plugin, which is not necessarily the case.

The stats plugin attempts to connect to the PVs `$(P)$(R)ProfileSizeX` and `$(P)$(R)ProfileSizeX_RBV` and the equivalent for `Y`.

The D8 Camera on I22 has the PVs `BL22I-DI-PHDGN-08:STAT:ProfileSizeX_RBV` and `BL22I-DI-PHDGN-08:STAT:ProfileSizeY_RBV`, but no writeable fields.

Having checked with Andy Smith on I22, changing this value would conventionally be achieved via changing region of interest.

## Acceptance Criteria
- The `NDPluginStatsIO` plugin no longer includes write PVs for the mentions non-writeable signals

DiamondJoseph, 2025-05-02:
As per the docs on the [ophyd-async NDStats plugin implementation](https://github.com/bluesky/ophyd-async/blob/6e6169cce2987f2e04e657c6ad197b01afdb7faf/src/ophyd_async/epics/adcore/_core_io.py#L63) pointing to the [documented fields on the AreaDetector plugin chain](https://areadetector.github.io/areaDetector/ADCore/NDPluginStats.html), is it the AreaDetector docs that need to be updated and the implementation adjusted, or is this an issue with the Stats plugin on i22's OAV?

Villtord, 2025-05-02:
I think it's the same for I09 analyser live view camera Stat Profiles - they are not writeable:

`(base) [eir17846@ws410 ~]$ cainfo BL09I-MO-ES-03:STAT1:ProfileSizeX_RBV
BL09I-MO-ES-03:STAT1:ProfileSizeX_RBV
    State:            connected
    Host:             172.23.240.13:5064
    Access:           read, no write
    Native data type: DBF_LONG
    Request type:     DBR_LONG
    Element count:    1
(base) [eir17846@ws410 ~]$ cainfo BL09I-MO-ES-03:STAT1:ProfileSizeX
Channel connect timed out: 'BL09I-MO-ES-03:STAT1:ProfileSizeX' not found.
`

Villtord, 2025-05-02:
Sorry I don't get your question - which is parent device?

In docs here https://areadetector.github.io/areaDetector/ADCore/NDPluginStats.html they say 

NDPluginStats Total
asynFloat64
r/o

it's read only and can be found either via 

$(P)$(R)Total_RBV if it is an "ai" record in epics

OR via

$(P)$(R)TotalArray if corresponding record in epics is of MCA type (I've never seen it before...)

Villtord, 2025-05-02:
Regarding the profile X size - it looks like it must be read/write according to docs, however beamline epics tells me it is just read.

coretl, 2025-05-09:
> As per the docs on the [ophyd-async NDStats plugin implementation](https://github.com/bluesky/ophyd-async/blob/6e6169cce2987f2e04e657c6ad197b01afdb7faf/src/ophyd_async/epics/adcore/_core_io.py#L63) [...]

Docs are wrong.

Correction is here: https://github.com/areaDetector/ADCore/pull/533

Anyone fancy submitting a PR to correct ophyd-async to make them read only?

DiamondJoseph, 2025-05-09:
@coretl I'm on it

PR #891 (2025-05-16): fix: correct signal types and types for Stats plugin
Fixes #881