bluesky/ophyd-async #884: Create the ability to retry signals

DominicOram, 2025-05-07:
We have a number of signals (specifically on the zebra but we have also seen some on the smargon) that, when set, will timeout occasionally. Sending the set again works and they don't timeout. Originally this was solved at the plan level but led to further issues https://github.com/DiamondLightSource/mx-bluesky/issues/1031.

Instead we should have the ability to specify that a signal is expected to potentially timeout and that we should re-set it if it does so. For example we could do this with a particular signal type:

```python
class RetrySignal(SignalRW):
    def __init__(retry_times: int, ...):
        ...

    async def set(...):
        number_of_retries = 0
        while(number_of_retries < self.retry_times):
            try:
                await super().set(...)
            except Timeout:
                logger.warn(...)
                number_of_retries += 1
```

However, the disadvantage of this is that it will be cumbersome for a user of `ophyd_async` to make a `Motor` retry without making an entirely new `RetryMotor` class. Similarly with any other `Device`.

## Acceptance Criteria
- There is a way to specify signals so that when they are set they will retry N times if there is a timeout on setting them
- When there is a timeout this should be logged
- As a nice to have it would be good to be able to do this on the `Motor` or `Device` level too

coretl, 2025-05-07:
Actually, could we put this logic in a wrapping function, then use it in a wrapping Device? 
```python
@AsyncStatus.wrap
async def set_with_retries(signal: SignalW[T], value: T, number_of_retries: int, timeout: float):
    """Set the signal value, retrying up to number_of_retries times in case of a TimeoutError"""

class Motor:
    def __init__(self, ..., number_of_set_retries: int = 0):
        self._number_of_set_retries = number_of_set_retries
        ...

    def set(self, value, timeout):
        status = set_write_retries(self.user_setpoint, value, self._number_of_set_retries, timeout)
        ...
```
Then if you need this in the zebra then you wrap it in a Device level `set` method?

DominicOram, 2025-05-07:
Are you suggesting we add this to the generic `Motor` then? I think it would be good to do so.

>Then if you need this in the zebra then you wrap it in a Device level set method?

The plans directly write to the signals though e.g. `abs_set(zebra.inp_1, 1)` so we would want a way to change every signal in the device so that it retries. This is also probably true of the `Motor`, we have seen instances where a set to `.VELO` will fail and need a retry.

coretl, 2025-05-09:
> This is also probably true of the `Motor`, we have seen instances where a set to `.VELO` will fail and need a retry.

Yuck, really? It hangs and times out? On which motor?

> The plans directly write to the signals though e.g. `abs_set(zebra.inp_1, 1)` so we would want a way to change every signal in the device so that it retries.

One more possibility, add an option for retry on failure to the `set` method in bluesky, then no changes to the device are needed, it's all at the plan level...

DominicOram, 2025-05-19:
>Yuck, really? It hangs and times out? On which motor?

Yep, on the smargon, see https://diamondlightsource.slack.com/archives/C048RD0KFFF/p1746534803627389?thread_ts=1746020339.267419&cid=C048RD0KFFF

> Discussing this, we decided it was best to push it down to the base signal, using https://stamina.hynek.me/en/stable/ or an API that looks like it

What would this look like to me if I:
* Explicitly don't want retries
* Have a single signal that I want to add more retries to
* Have a device I want to add retries to every signal on

coretl, 2025-05-21:
I imagine we would add `retries=0` to all the `epics_signal_...` functions

> What would this look like to me if I:
> 
>     * Explicitly don't want retries

Do what you do at the moment

>     * Have a single signal that I want to add more retries to

```python
sig = epics_signal_rw(float, prefix+"foo", retries=3)
```

>     * Have a device I want to add retries to every signal on

```python
sig1 = epics_signal_rw(float, prefix+"foo", retries=retries)
sig2 = epics_signal_rw(float, prefix+"foo2", retries=retries)
```

Although I would probably say for something like a motor you'd do the retries yourself in the `set()` method

closed 2025-06-30 by shihab-dls

DominicOram, 2025-07-01:
As an FYI, we haven't seen many repeats of these issues recently so will not immediately start using this feature. However, it's one of those that's will be useful to have in our back pocket on a weekend support call. I will try and let you know when we use it and on what PVs.

PR #943 (2025-06-30): Add retries to signals
Fixes #884 

This uses `stamina` (https://stamina.hynek.me/en/stable/) to retry setting to signals.