bluesky/ophyd-async #885: Describe how ophyd-async implements bluesky verbs

coretl, 2025-05-07:
Document the following, and expand for the other verbs:
- prepare: provide the parameters and prepare for an upcoming scan, like exposure time and number of frames for a detector including arming if hardware triggered, or the start and end position for a constant velocity scan on a single motor that moves it to the run up position
- set: ask the device to go to a given setpoint in a software step scan
- trigger: ask a detector to take data at a single step of a software step scan
- kickoff: start a flyscan, for a detector this does nothing if hardware triggered, for a motor this start the motion and returns when at velocity
- complete: wait until the flyscan is done, for a detector this waits for the frames to be written, for a motor this waits for the ramp down to finish

## Acceptance Criteria
- Docs exist

stan-dot, 2025-05-28:
this question was raised in a SWIFT meeting, scientists need clarity if they are to write plans themselves soon

stan-dot, 2025-05-28:
found this https://github.com/bluesky/ophyd-async/issues/905