bluesky/ophyd-async #894: Decide once and for all about how Devices are named

coretl, 2025-05-14:
At the moment, Device naming happens via `Device.set_name`. This is called:
- in `Device.__init__(name)` at construction
- after construction by the `init_devices` or `dodal.common.beamlines.beamline_utils.device_factory`

This leads to some odd edge conditions:

```python
class Base(Device):
    def __init__(self, name=""):
        self.signal1 = soft_signal_rw()
        super().__init__(name=name)

class Derived(Base):
    def __init__(self, name=""):
        super().__init__(name=name)
        self.signal2 = soft_signal_rw(name="blah")

# Naming at init only names those that are present when the superclass init is called
derived1 = Derived(name="derived")
assert derived1.signal1.name == "derived-signal1"
assert derived1.signal2.name == "blah"

# Naming after init names everything
derived2 = Derived()
derived2.set_name("derived")
assert derived2.signal1.name == "derived-signal1"
assert derived2.signal2.name == "derived-signal2"
```
This is inconsistent, so we should fix it.

Possible solutions:
1. Remove `name` from `Device.__init__` and require `Device.set_name` to be called after creation
2. Set the name of child devices in `Device.__setattr__`

Both of those solutions will break one particular @rtuck99 had (that I can't remember details of) where `set_name` was never used after init, so the name "blah" in the above example was actually the desired name.

To support that use case we would have to override `Derived.set_name` to explicitly name `signal2` to be "blah", either by overriding the baseclass `set_name` (for 1):

```python
class Derived(Base):
    def set_name(self, name: str, *, child_name_separator: str | None = None) -> None:
        super().set_name(name, child_name_separator=child_name_separator)
        self.signal2.set_name("blah")
```

or by making an intermediate function that could be overridden instead (for 2):

```python
class Derived(Base):
    def set_child_name(self, child, name):
        super().set_child_name(child, "blah" if child is self.signal2 else name)
```

I think 2 would cause less breakages, so I would favour that over 1, but I'd welcome thoughts on the best API to override the child name.

@DominicOram @DiamondJoseph @callumforrester @jwlodek thoughts?
        
## Acceptance Criteria
- ADR written
- Code changed
- Only the single test that mimics @rtuck99's use case is changed

callumforrester, 2025-05-16:
How about a flag in `set_name`?

```python
derived1 = Derived()
derived1.set_name("derived", recursive_override=True)
assert derived1.signal1.name == "derived-signal1"
assert derived1.signal2.name == "blah"

derived2 = Derived()
derived2.set_name("derived", recursive_override=False)
assert derived2.signal1.name == "derived-signal1"
assert derived2.signal2.name == "derived-signal2"
```

coretl, 2025-05-19:
Go with approach 2, then look at how expensive `set_name` actually is when called repeatedly

rtuck99, 2025-05-19:
@coretl I've had a look through the code and I don't seem to be able to find the example case we discussed previously. I'm not sure I can think of any particularly good reason why we would need special naming for sub-devices. I suspect we have probably refactored that code in the meantime.
It may have been something we did because some sub-devices were dual-parented, a thing we now don't have to do because we can declare opaque references.

coretl, 2025-05-19:
Turns out that `set_name` when the name is already set is almost never called, so this is a reasonable approach.

commit 2f148d8e (2025-06-27): Set device name when child device attached (#899)

* Set device name when child device attached

Fixes #894

* Mention when naming is added

* Add docs
closed 2025-06-20 by coretl