bluesky/ophyd-async #903: Extend DeviceVector to support string and integer

oliwenmandiamond, 2025-05-27:
In  the ophyd-async documentation, there is the DeviceVector class:
"
Sometimes, it makes sense to group devices by number, say an array of sensors:

```Python
class SensorGroup(StandardReadable):
    def __init__(self, prefix: str, name: str = "", sensor_count: int = 3) -> None:
        with self.add_children_as_readables():
            self.sensors = DeviceVector(
                {i: Sensor(f"{prefix}{i}:") for i in range(1, sensor_count + 1)}
            )

        super().__init__(name)
```

DeviceVector allows writing maintainable, arbitrary-length device groups instead of fixed classes for each possible grouping. A DeviceVector can be accessed via indices, for example: my_sensor_group.sensors[2]. Here sensors is a dictionary with integer indices rather than a list so that the most semantically sensible indices may be used, the sensor group above may be 1-indexed, for example, because the sensors’ datasheet calls them “sensor 1”, “sensor 2” etc. 
"

However, it would also be useful to support str as well as int as the key for DeviceVector e,g

```Python
class ExcitationEnergySources(StandardReadable):
    def __init__(self, name: str = "", alias_name_to_device: dict[str, Reference[Device]]) -> None:
        with self.add_children_as_readables():
            self.sources = DeviceVector(
                alias_to_device
            )

        super().__init__(name)
```
```Python
pgm = PGM(...)
dcm = DCM(...)
energy_sources = ExcitationEnergySources("energy_sources", {"source1" : Reference(DCM), "source2" : Reference(PGM)})

#Now have a way to dynamically get excitation energy source for a region from a plan

def plan(region: AbstractBaseRegion, energy_sources = inject("energy_sources")) -> Device:
     source_alias : str = region.excitation_energy_source
     return energy_sources.sources[source_alias]
```

coretl, 2025-06-06:
In ophyd-async at the moment we have 2 ways to make a tree of Devices where their read output is a tree of the children:
- `StandardReadable` if the children have string names
- `DeviceVector` if the children have integer names

I'm loath to extend `DeviceVector` to support arbitrary typed keys, as I think this will muddy the waters between the 2 use cases.

In your second use case I can't see why you need `DeviceVector` rather than just a plain `dict`. The only reason to make it a `DeviceVector` is so you can support `read()` and child naming. 

In your first use case I would like to see how you use `ExcitationEnergySources`, as it appears you are escaping the naming using `Reference`, so only need `read()`. For this case you can make an arbitrary `sources = StandardReadable()`, then call `sources.add_readables()` from the outside.

coretl, 2025-06-09:
> `{'pgm-energy': {'value': 0.0, 'timestamp': 1019825.480868053, 'alarm_severity': 0}}`
> 
> I need to map the alias name of source1 or whatever I call it to pgm-energy. A DeviceVector allows me to do exactly that, StandardReadable does not.

Ok, so in this case you would like it to produce:
`{'source1': {'value': 0.0, 'timestamp': 1019825.480868053, 'alarm_severity': 0}}`

But if you do `pgm.read()` then would you still like the original output?
`{'pgm-energy': {'value': 0.0, 'timestamp': 1019825.480868053, 'alarm_severity': 0}}`

If so, then that is not possible with `DeviceVector` either. `ophyd-async` will have precisely one name for a signal, and will always put that into the `read()` dictionary no matter whether it is read from the originating device, or via an alias or `Reference`.