bluesky/ophyd-async #936: Allow for specifying a different write and read location for detector writers

jwlodek, 2025-06-20:
In certain cases (most commonly windows detectors that have a drive letter in the path name), we need to be able to write to one location, and then read the file back from another location. In ophyd sync, this is done with `read_path_template` and `write_path_template`.

We should provide something similar for ophyd-async. I propose the following:

* `PathInfo` gets a new `read_path` optional attribute that defaults to `None`
* Existing providers are modified to add a `read_path` kwarg that defaults to None, that is passed through to the `PathInfo` returned by `__call__`
* Detector writers are modified to have the resource doc generated be the read path if not None, otherwise retain existing behavior.

Thoughts? @coretl 

## Acceptance Criteria
- Have option to specify different read/write locations for detector writers.
- Unit testing covers new feature.
- All existing `PathProvider` classes and `DetectorWriter` classes modified to support this.

coretl, 2025-06-20:
The existing `PathInfo` takes https://github.com/bluesky/ophyd-async/blob/9f724d25e39b1be30f6441566a8153acd3388e77/src/ophyd_async/core/_providers.py#L21-L23

I suggest that we:
- add a new `directory_uri: str` to `PathInfo` that defaults to the code in https://github.com/bluesky/ophyd-async/blob/9f724d25e39b1be30f6441566a8153acd3388e77/src/ophyd_async/core/_hdf_dataset.py#L57-L66 without the filename
- existing `PathProviders` modified to optionally allow a different `directory_uri` to be passed in
- detector writers updated to use `directory_uri` rather than `directory_path` when generating resources

jwlodek, 2025-08-20:
Yep, marking as completed.

closed 2025-08-20 by jwlodek