bluesky/ophyd-async #976: AreaDetector trigger timing out in system tests

DiamondJoseph, 2025-07-23:
ADSim timing out on stage/trigger in system tests in ophyd-async 0.12.x

```
<AsyncStatus, device: det, task: <coroutine object StandardDetector.stage at 0x7faca83c6340>, errored: TimeoutError("det-driver-acquire didn't match False in 10.0s, last value True")>
ERROR system_tests/test_adsim.py::test_plan_produces_expected_stop_document[1-1] - bluesky.utils.FailedStatus: <AsyncStatus, device: det, task: <coroutine object StandardDetector.trigger at 0x7facd2e8c040>, errored: TimeoutError("Timeout Error while waiting 10.001s to update ['ca://BL01T-DI-CAM-01:HDF5:NumCaptured_RBV']. Last observed signal and value were (<ophyd_async.core._signal.SignalR object at 0x7facd2ec4790>, 0)")>
ERROR system_tests/test_adsim.py::test_plan_produces_expected_descriptor[1] - bluesky.utils.FailedStatus: <AsyncStatus, device: det, task: <coroutine object StandardDetector.trigger at 0x7facd2f45140>, errored: TimeoutError("Timeout Error while waiting 10.001s to update ['ca://BL01T-DI-CAM-01:HDF5:NumCaptured_RBV']. Last observed signal and value were (<ophyd_async.core._signal.SignalR object at 0x7facd2f2c790>, 0)")>
```

## Steps To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Acceptance Criteria
- Specific criteria that will be used to judge if the issue is fixed

jwlodek, 2025-07-23:
I can't seem to reproduce with my test camera:

```
In [14]: ophyd_async.__version__
Out[14]: '0.12.2'

In [15]: RE(count([manta2], 1))
========= Emitting Doc =============
name = 'start'
{
    "uid": "395292ee-f9ce-462a-a140-109e5d36655b",
    "time": 1753277525.3011668,
    "cycle": "2025-2",
    "data_session": "pass-56789",
    "versions": {
        "ophyd": "1.10.7",
        "ophyd_async": "0.12.2",
        "bluesky": "1.14.2"
    },
    "scan_id": 5,
    "plan_type": "generator",
    "plan_name": "count",
    "detectors": [
        "manta2"
    ],
    "num_points": 1,
    "num_intervals": 0,
    "plan_args": {
        "detectors": [
            "<ophyd_async.epics.advimba._vimba.VimbaDetector object at 0x7f271879fd90>"
        ],
        "num": 1,
        "delay": 0.0
    },
    "hints": {
        "dimensions": [
            [
                [
                    "time"
                ],
                "primary"
            ]
        ]
    }
}
============ Done ==================
========= Emitting Doc =============
name = 'descriptor'
{
    "configuration": {
        "manta2": {
            "data": {
                "manta2-driver-acquire_period": 0.014692,
                "manta2-driver-acquire_time": 0.1
            },
            "timestamps": {
                "manta2-driver-acquire_period": 1753277365.030311,
                "manta2-driver-acquire_time": 1753277364.908343
            },
            "data_keys": {
                "manta2-driver-acquire_period": {
                    "dtype": "number",
                    "shape": [],
                    "dtype_numpy": "<f8",
                    "source": "ca://XF:31ID1-ES{GigE-Cam:2}cam1:AcquirePeriod_RBV",
                    "units": "",
                    "precision": 3
                },
                "manta2-driver-acquire_time": {
                    "dtype": "number",
                    "shape": [],
                    "dtype_numpy": "<f8",
                    "source": "ca://XF:31ID1-ES{GigE-Cam:2}cam1:AcquireTime_RBV",
                    "units": "",
                    "precision": 3
                }
            }
        }
    },
    "data_keys": {
        "manta2": {
            "source": "ca://XF:31ID1-ES{GigE-Cam:2}HDF1:FullFileName_RBV",
            "shape": [
                1,
                272,
                364
            ],
            "dtype": "array",
            "dtype_numpy": "|u1",
            "external": "STREAM:",
            "object_name": "manta2"
        }
    },
    "name": "primary",
    "object_keys": {
        "manta2": [
            "manta2"
        ]
    },
    "run_start": "395292ee-f9ce-462a-a140-109e5d36655b",
    "time": 1753277526.3347414,
    "uid": "208d7ff1-5cf5-4a96-a58a-ddedb950e712",
    "hints": {
        "manta2": {
            "fields": [
                "manta2"
            ]
        }
    }
}
============ Done ==================
========= Emitting Doc =============
name = 'stream_resource'
{
    "uid": "1f9948b1-bbd2-4f94-a5c7-e541b09c5eca",
    "data_key": "manta2",
    "mimetype": "application/x-hdf5",
    "uri": "file://localhost/nsls2/data/tst/legacy/mock-proposals/2025-2/pass-56789/assets/manta2/2025/07/23/JAZ7KsytzDYWLbex3JAKSZ.h5",
    "parameters": {
        "chunk_shape": [
            1,
            272,
            364
        ],
        "dataset": "/entry/data/data"
    },
    "run_start": "395292ee-f9ce-462a-a140-109e5d36655b"
}
============ Done ==================
========= Emitting Doc =============
name = 'stream_datum'
{
    "stream_resource": "1f9948b1-bbd2-4f94-a5c7-e541b09c5eca",
    "uid": "1f9948b1-bbd2-4f94-a5c7-e541b09c5eca/0",
    "seq_nums": {
        "start": 1,
        "stop": 2
    },
    "indices": {
        "start": 0,
        "stop": 1
    },
    "descriptor": "208d7ff1-5cf5-4a96-a58a-ddedb950e712"
}
============ Done ==================
========= Emitting Doc =============
name = 'event'
{
    "uid": "f4b7fe7e-37e1-4682-92d3-cbceba4c195a",
    "time": 1753277526.3522973,
    "data": {},
    "timestamps": {},
    "seq_num": 1,
    "filled": {},
    "descriptor": "208d7ff1-5cf5-4a96-a58a-ddedb950e712"
}
============ Done ==================
========= Emitting Doc =============
name = 'stop'
{
    "uid": "24e3fc22-cd14-4b71-9612-0d7c613a5044",
    "time": 1753277526.3528042,
    "run_start": "395292ee-f9ce-462a-a140-109e5d36655b",
    "exit_status": "success",
    "reason": "",
    "num_events": {
        "primary": 1
    }
}
============ Done ==================
Out[15]: ('395292ee-f9ce-462a-a140-109e5d36655b',)
```

Let me look at the system tests to see if they are doing something different.

coretl, 2025-07-29:
This is https://github.com/DiamondLightSource/aioca/issues/68

closed 2025-07-31 by shihab-dls