bluesky/ophyd-async #984: Area detector NDAttributes do not produce stream_datum keys when being collected

canismarko, 2025-07-29:
I have an area detector with several NDAttributes [set up during staging](https://github.com/spc-group/haven/blob/66f842ce701f1478f0f6e39a8135de87aba92de7/src/haven/devices/detectors/xspress.py#L248) as XML written directly to the NDAttributes PV. The cam is also [included as a plugin](https://github.com/spc-group/haven/blob/66f842ce701f1478f0f6e39a8135de87aba92de7/src/haven/devices/detectors/xspress.py#L220) to ensure the NDAttributes are included in the descriptor.

The detector is then used in [a fly scan](https://github.com/spc-group/haven/blob/66f842ce701f1478f0f6e39a8135de87aba92de7/src/haven/plans/_fly.py#L274), where is is staged, then prepared (`bps.prepare(detector)`), a stream is declared (`bps.declare_stream(detector)`), and the detector is then kicked-off, completed, and collected.

Here is the exception that was raised, and the documents that were emitted up until the exception: https://gist.github.com/canismarko/b1cb9af37c5c87362be0c454967f000f

This problem does not happen when using e.g. ``bp.count(detector)``.

## Steps To Reproduce
Steps to reproduce the behavior:
1. Create an area detector with ndattributes on the camera, set up so that these ndattributes are included in the descriptor document.
2. Execute a fly scan, include a collect.
3. See error

## Acceptance Criteria

The area detector above should complete the fly scan, collect its data, and produce documents with the correct data.

canismarko, 2025-07-29:
I think I may have figured out why this is happening.

During `ophyd_async.epics.adcore._hdf_writer.ADHDFWriter.open()`, the following steps are executed, in order:

1. `self._datasets` is created with a single entry for the NDArray.
2. `self._composer` is created and passed `self._datasets` as an argument.
3. The NDAttributes XML is read and additional HDFDatasetDescription() objects are added to `self._datasets`.

During `HDFDocumentComposer.__init__()`, it iterates of `datasets` to produce `self._bundles()`. However, since step hasn't happened yet, the NDAttributes are not included in the bundles, just the NDArray dataset.

If I reorganize `ADHDFWriter.open()` so that step 2 is after step 3, then `RE(bp.count(detector))` executes successfully and produces the following documents: https://gist.github.com/canismarko/0c35a45d343f2d88bd62e0e9191108b4#file-documents

Alternately, HDFDocumentComposer could be rewritten so that `self._bundles` is created the first time `HDFDocumentComposer.make_stream_docs()` is called.

coretl, 2025-08-01:
Good catch! Yes, please reorder 2 and 3. Are you happy to make a PR for this?