bluesky/ophyd-async #1009: Allow SimMotor to set initial value when creating

oliwenmandiamond, 2025-08-11:
When creating a SimMotor for a test and I want the motor to be a specific value, I have to do this

```Python
@pytest.fixture
async def pgm(RE: RunEngine) -> SimMotor:
    pgm = SimMotor("pgm", instant=True)
    await pgm.set(100)
    return pgm
```

I should be able to do this with one line

```Python
@pytest.fixture
async def pgm(RE: RunEngine) -> SimMotor:
    return SimMotor("pgm", instant=True, initial_value=100)
```

## Acceptance Criteria
- There's an additional argument to set the initial value

oliwenmandiamond, 2025-08-12:
It would also be great if we could optionally specify the units for the sim motor too.

```Python
SimMotor("pgm", instant=True, initial_value=100, units="eV")
```

fajinyuan, 2025-08-27:
Hi @coretl @oliwenmandiamond could you please assign this issue to me so I can start to work on this please?

coretl, 2025-09-01:
You can always fork and make a PR, but I have invited you to the org so you can push to a branch of the repo.