bluesky/ophyd-async #1039: Provide the ability to inject logic when mock devices are created

DominicOram, 2025-09-05:
We have a number of devices where for a mock of the device to at all be useful we need to inject some logic. The obvious example is something like:

```python
class Device():
    def __init__(self, prefix):
        self.setpoint = epics_signal_w(prefix+":SP")
        self.readback = epics_signal_w(prefix+":RBV")
```

where to make a mock of the device behave like expected we need to do something like:

```python 
with init_devices(mock=True):
    my_device = Device()

callback_on_mock_put(my_device.setpoint, lambda _: set_mock_value(my_device.readback))
```

This works fine for individual tests but it has some disadvantages:
* We inevitably need the same logic for tests of the device itself and the plans. But plans/devices are normally in separate repos for us. This leads to arguments about where the logic should go (https://github.com/DiamondLightSource/dodal/issues/1455)
* It can be useful on a beamline to be able to quickly switch a device to mock mode. The usual usecase for this is that hardware has failed but to get us close to operational quickly we want to run the same plans as usual just with mock data. In some cases the logic above is required to get to a point where the plan will run without failure.
* Personal preference but I think it's generally cleaner to have the device logic and the "expected ioc logic" (which is essentially what we're writing here) close to each other

I propose that `ophyd-async` should give us the ability to add this behaviour automatically any time the device is instantiated with `mock=True` e.g. some kind of interface like:

```python
class Device():
    def __init__(self,prefix):
        self.setpoint = epics_signal_w(prefix+":SP")
        self.readback = epics_signal_w(prefix+":RBV")
 
   def _default_mock_logic(self):
       callback_on_mock_put(my_device.setpoint, lambda _: set_mock_value(my_device.readback))
```

I guess some that there may be times you don't want that behaviour on `mock=True`. But I think this can be handled on a case by case basis. E.g. if you want to test what happens to the above device when data isn't getting passed through from `setpoint` -> `readback` then our test can just remove the callback as it's first thing.

## Acceptance Criteria
- It is easier to specify logic for mock devices alongside each device.

coretl, 2025-10-23:
As discussed in person, I agree with adding some generally useful mock behaviour, but we want to be able to opt out of it so tests can implement their own. I will have a think about the best mechanism to support this

coretl, 2025-11-11:
A suggestion for implementation, add this via a subclass to `LazyMock` (which we should rename to `DeviceMock`). Give the baseclass a sync `connect` method, which by default does nothing, then allow it to be overridden like so:

```python
@default_device_mock_for_class
class InstantMotorMock(DeviceMock[Motor]):
    def connect(self, device: Motor):
        callback_on_mock_put(device.setpoint, lambda v: set_mock_value(device.readback, v))
```
The `default_device_mock_for_class` (suggestions for a better name welcome) decorator registers this `DeviceMock` subclass as the default one for the type of `Device`, in this case `Motor`.

Then when `Motor.connect(mock=True)` is called, then `Device.connect` should create an instance of the correct `DeviceMock`for that class, falling back to a base `DeviceMock` if one hasn't been produced. 

If we want different behaviour to the default, then pass a different `DeviceMock` to the connect method, e.g.:
```python
await motor.connect(mock=ErrorMovingMotorMock(time_to_fail=2))
```

@DominicOram does this work for you?